seehase/ZonPHP #3: Ignore missing CSV Files

qutschwalze, 2018-04-23:
I have some gaps in my CSV files. Is it possible to ignore the missing CSV. Now it seems that ZonPHP is stopping if a CSV is missing. 

Thank you.

seehausen, 2018-04-23:
Which importer do you use?
do you have all your "old" CSV files so is it possible refresh all the data?

This could happens if you have a gap for more than 60 days
in this case you can increase the number of days

search in the used importer for 
`$adag = array();
for ($tel = 0; $tel <= 60; $tel++) {`

and increase the number of 60 to the max amount of days that are missing

to refresh the data goto 
<yourwebsite>/install/clear_tables.php
then all data will be delete, after going back to the index page the data will be imported again... this will take some time and some page reloads

qutschwalze, 2018-04-23:
Wow, yes that worked. 
Thank you!

seehase, 2018-04-23:
OK than I close this issues

closed 2018-04-23 by seehase