seehase/ZonPHP #4: Invullen_gegevens_sunny_explorer_multi.php doenst work

qutschwalze, 2018-04-24:
I have two SMA Powerstations which creates a combined CSV file. 
In the old Version from zonPHP there was a hack for more than one Powerplant.
I assume that the Invullen_gegevens_sunny_explorer_multi.php is the original hack, but if I activate the importer, I got this failure:

Query failed. ERROR: You have an error in your SQL syntax; check the manual that corresponds to your MariaDB server version for the right syntax to use near '' at line 1

seehase, 2018-04-24:
Can you please send me some sample CSV files than I will have a look
I don't know the "hack" so what is your expectation? having one chart which shows the sum of both converter or having two seperate chart for each converter?

Please send me the files to "solar@seehausen.org" I speak german ;-)

seehausen, 2018-10-30:
Is this topic still open? If yes please provide me some sample CSV files

seehase, 2018-11-09:
no feedback , i close the issue

closed 2018-11-09 by seehase