seehase/ZonPHP #5: multi inverter chart 'all years overview' not working

nthijssen, 2018-10-30:
As I have multiple inverters logging onto ZonPHP, this issue shows up:
When having installed latest version (v2018.01.04), I open from main screen the chart 'All years overview' (left column, third chart from the top).

**As actual result,** the chart shows all years, but from ALL inverters, while 1 inverter(name) has been selected.

**As expected result**, the chart should show all years from selected inverter (visible in top of the screen of ZonPHP).
Switching between inverter / all inverters buttons won't solve it. Possibly an issue in query part.

seehausen, 2018-10-30:
did you checked the current development version v2018.06.02
https://github.com/seehase/ZonPHP/archive/development.zip

I did some changes. Please have a look if your issue is still there

nthijssen, 2018-10-30:
I did download your development set, and installed it. Sadly, also in v2018.06.02 it is still having the same issue.

seehase, 2018-11-01:
Hi Niels
I'm not sure that I understand the bug, can you send me some screenshots
and is you website available in the internet (you can send me a private mail to zonphp@seehausen.org)

Can you reproduce the bug on my site https://solar.seehausen.org I also have 2 inverters

Please provide me some more information, then I try to fix it
Kind Regards
Holger

seehase, 2018-11-01:
but the 
https://www.renewable-energy-now.org/zonphp7/all_years_overview.php?naam=JIANGSUKINGSUN
looks different as
https://www.renewable-energy-now.org/zonphp7/all_years_overview.php?naam=SUNTECH

I don't see the problem

nthijssen, 2018-11-04:
[image]
Above an overview of inverter/system MSK, of all years showing in ZonPHP v2.25 version.

But in the development version just downloaded, this chart can't be reproduced (yet):
For MSK selected system it shows (despite toggling between inverter/all inverters):
[image]

seehase, 2018-11-04:
OK, I see the problem 
I will have a look

seehase, 2018-11-09:
Hi @nthijssen 
I think I solved the problem 
please install current developer version and provide feedback if problem still exists

nthijssen, 2018-11-11:
Tested on build 2018.11.06 and it works now! Pass!

Niels Thijssen
nielsthijssen75@gmail.com <mailto:nielsthijssen75@gmail.com>

My OpenPGP key is: 75FD3735

closed 2018-11-11 by seehase