seehase/ZonPHP #7: daily chart: toggle button between selected inverter and all inverters

nthijssen, 2018-10-31:
Build: v2018.06.02

The daily chart shows selected inverter.
But in situation that one has more than 1 inverter, it would be usefull to have a toggle button on the daily chart. The toggle button would switch the chart betwwen 1 (selected) inverter, and ALL inverters.
The chart would be preferably using different colors, a summarized production chart. So all power output and production data is added on top each other.

End user can than have a tool to predict when it's optimum to turn on local appliences, to take benefit of the sunshine.

nthijssen, 2018-11-04:
> Hi @nthijssen
>
> I implemented a first version of a dropdown Button in the day chart
> you can see the results here
>
> https://solar.seehausen.org/day_overview.php
>
> Is this how you want to have it?
>
> we can think of placing the dropdown to another position, and of course color
> If we agree, I will put the dropdown to the other charts as well
>
> Open points
>
> I have to fix is if you had choosen another day than today, if you click the button, it will go back to today... I'm working on it
> add "All Inverters" to the list as well
> You can try yourselfe, if you install the current developer version from github
>
> Please let me kwow what you think and what I should improve
>
> Waiting for your Feedback
>
> Kind regards
> Holger

Will try to test and share feedback today

Verstuurd vanaf mijn iPhone

nthijssen, 2018-11-04:
Hello,

I downloaded just a moment ago the development version and installed it in https://www.renewable-energy-now.org/zonphp71 <https://www.renewable-energy-now.org/zonphp71>
When going to the daily overview, I see the drop down list of the available inverters. This looks good.

But whether selecting a different inverter or selecting ‘ all inverters’ , will result in an error.
it drops back to my website, with a sort of http 404  error, can’t be found etc.

Niels Thijssen
nielsthijssen75@gmail.com <mailto:nielsthijssen75@gmail.com>

My OpenPGP key is: 75FD3735

seehase, 2018-11-04:
@nthijssen 
I just fixed this bug
please update the file day_overview.php

seehase, 2018-11-09:
@nthijssen 
I implemented the togle button in all charts

closed 2018-11-09 by seehase

nthijssen, 2018-11-11:
Retested in Build 2018.11.04, passed/approved!

Niels Thijssen
nielsthijssen75@gmail.com <mailto:nielsthijssen75@gmail.com>

My OpenPGP key is: 75FD3735