seehase/ZonPHP #12: add line with max. production to chart daily view

nthijssen, 2018-11-11:
Tested with build v2018.11.04 today (11-nov).

As end-user I'd like to see on viewing daily and monthly charts (of one or more inverters) to see the date (dd-mm-yyyy) and the amount (in kWh) of the maximum production registered ever.
This will help me evaluate when my system is approaching a ' best ever'  production moment, and maybe debug issues with my system.

Attached is a screenshot of PHP 2.25 where I have added this, as example in orange color highlighted:
[image]

If you like, I can share the code I have used.

seehase, 2018-11-11:
I will add that date and also the date in the popup of the chart... it was on my list for a long time

seehase, 2018-11-11:
fixed it in Version build 2018.11.05
please clear sesseion afer installing new Version

nthijssen, 2018-11-11:
Tested with build 2018.11.05, with clearing of cache/session.

I see that now the best day of production in CURRENT month is shown.
What I would like to see, is the best day of production of current month in **history**.
E.g.:
in November 6h, 2017 it was measured (either combined (all inverters) or as single selected inverter, as .e.g. 12,41kWh

Now it shows in current month the maximum production. Hope this is clearly described?

seehase, 2018-11-11:
I'm not sure
https://solar.seehausen.org/day_overview.php?naam=SEEHASE
the max day for november in history is the 1.11.2015, with 23.54 kWh and max peak 2660h
http://127.0.0.1/ZonPHP/ZonPHP/day_overview.php?dag=11/01/2015

This is also the gray max line... do you want do see something else

seehase, 2018-11-11:
@nthijssen the best day of the current month can be seen in month_overview
https://solar.seehausen.org/month_overview.php 

I like to see the max day in month for all years like the max curve in the chart
do you agree?
btw: I changed the max date to a link that you can directy jump to that day by clicking

closed 2018-11-11 by seehase