seehase/ZonPHP #18: Enhance day view to show more than one invertert at once

seehase, 2019-03-08:
As proposal from Eric, we can change day chart to show all inverter in one view with posiblitiy to swicht between converters

[image]
the chart is additiv

ericplan, 2019-03-12:
Have a working demo on http://www.e2fb.nl/zphp/day_twee.php. All work is done in the daycharts file (line 238 - 253). Output from the loop is strict JSON and can be passed directly to JS. But I have to work on the correct data for the selection (also on the max and cumulative lines). Also work on creating the javascript variables for every inverter in a for loop and add them to the Highcharts graph automatically. For now I created them by hand.

seehase, 2019-03-12:
Can you attach the code of your working sample, then I try to integrate it

ericplan, 2019-03-12:
Hi Holger,

Here you are. Be extremely careful with the foreach output. Hard to get it strict JSON, but it pays off because you can use it immediately in the Highcharts series.

Regards,

Eric

ericplan, 2019-03-22:
Since this implemented in the new development build and after some thinking about it, why not implement the "max"- and "cum" line 'stacked' as well? Should be nice.

seehase, 2019-03-23:
Finished day view, continue working also an month/year and all year charts ... stay tuned

seehase, 2019-03-27:
added multiview for months as well, year overview will follow

seehase, 2023-05-13:
Already done

closed 2023-05-13 by seehase