seehase/ZonPHP #20: multiple inverters - day view - differentiate / use more colours

nthijssen, 2019-03-19:
when a day-view is selected, with multiple inverters, there's one coloured area visible with the sum of the 'n' inverters. Would be nice to have (in case of 3 inverters/systems), 3 shades of color visible to differentiate more clearly. Is this possible?
I use currently the Development version v2018.11.14.

seehausen, 2019-03-19:
Hi Niels
I already worked on this, see
https://solar.seehausen.org/day_overview.php?type=all

I checked in the code on sunday, but did not increased the build number so far
please checkout the lates development branch, and I would like to get your feedback

Currently I have to change the max line and cum line according to all inverter

nthijssen, 2019-03-19:
downloaded dev-version 2018.11.05 and installed. indeed now multi-inverter show separate colors. Thanks. Issue can be closed.

seehase, 2019-03-20:
Will be fixed with [Issus#18]

closed 2019-03-20 by seehase