seehase/ZonPHP #21: Enable full screen mode on ios

ericplan, 2019-03-20:
IPhone and iPad are capable of running in full screen mode by adding a simple meta tag in the page header <meta name="apple-mobile-web-app-capable" content="yes">

This can be placed in the header.php. The result is shown in the pictures
[image]
[image]

seehausen, 2019-03-20:
Hi Eric
can you discribe the metatag which has to be set? then I can add that

ericplan, 2019-03-20:
I added it to the message, for some reason it disappeared after inserting the pics.
<meta name="apple-mobile-web-app-capable" content="yes">

ericplan, 2019-03-20:
`<meta name="apple-mobile-web-app-capable" content="yes">`

seehase, 2019-03-20:
I commit and pushed the change to the header.php
can you checkout latest dev branch and check? I can test the results this evening too.

ericplan, 2019-03-20:
It is in the same place where i putted it, so i guess it works. 

BTW There is a typo in the word “monitoring” on line 18

seehase, 2019-03-22:
typo fixed

closed 2019-03-22 by seehase