seehase/ZonPHP #25: Diary / Tagebuch in Developer layout issue

nthijssen, 2019-04-11:
[image]

Diary or events registered in tagebuch.txt (with HTML markup) are shown on screen with incorrect layout. Fix needed (most likely) in file mydiary.php.

Detected since v2018.11.15 (dev)

seehausen, 2019-04-12:
@nthijssen  Can you upload your tagebuch.txt for testing please?

seehase, 2019-04-12:
HI @nthijssen 
I fixed the mydiary in v2019.03.13 (dev)
please checkout latest dev branch and provide feedback  (only mydiary.php was changed ... and versioninfo"

Kind regards
Holger

nthijssen, 2019-04-13:
[tagebuch.txt](https://github.com/seehase/ZonPHP/files/3075991/tagebuch.txt)
see attached tagebuch.txt as requested.

nthijssen, 2019-04-13:
Tested on/with build 2018.11.15 (dev), this issue is now solved. Retest: passed.

closed 2019-04-13 by nthijssen