seehase/ZonPHP #65: No import of sunny explorer csv to database

bertl85, 2023-08-14:
Hello

I just installed the newest version, but the values from the csv will noch be imported in the database.
I cannot find any failure in the parameters.
The old version worked well and if I import the values with the old version, the values are also shown in the new one.

pv.regensburg-wetter.de

seehausen, 2023-08-14:
Hi
is it possible to send me one or two of your *.csv files
and the some of your parameters to solar@seehausen.org
* plantnames = 
* importer = 
* [yourPlant]
  importPrefix = solar1
and the path where your csv files are located

seehase, 2023-08-19:
Bug should be fixed with latest release
https://github.com/seehase/ZonPHP/releases/tag/Release2023.08.19

kds70, 2023-08-25:
At first - a nice peace of software. I still love it 👍 

But sure it´s fixed with Release2023.08.19 ?
The same issue as @bertl85 here.
It still doesn´t import with sunny_explorer.php - if i use the old one "Invullen_gegevens_sunny_explorer.php" it imports the day values but not the months and the db values are zero.
With v2023.06.01 and Invullen_gegevens_sunny_explorer.php it works well.
Thank you -

seehase, 2023-08-25:
> Zunächst einmal eine schöne Software. Ich liebe es immer noch 👍
> 
> Aber sicher, dass es mit Release 2023.08.19 behoben ist? Das gleiche Problem wie@bertl85Hier. Es wird immer noch nicht mit sunny_explorer.php importiert. Wenn ich das alte „Invullen_gegevens_sunny_explorer.php“ verwende, werden die Tageswerte importiert, aber nicht die Monate, und die Datenbankwerte sind Null. Mit v2023.06.01 und Invullen_gegevens_sunny_explorer.php funktioniert es gut. Danke -

Könntest du mir vielleicht auch ein paar deiner CSV Dateien schicken und einen Link auf deine Seite falls die öffentlich ist?
gerne privat an solar@seehausen.org
Wenn deine Seite nicht öffentlich, dann bitte die paramerters.php schicken vorher deine Passwörter entfernen!

kds70, 2023-08-25:
Hallo @seehase - mach ich. Die Seite ist öffentlich:
http://pv-scherbaum.de/index.php
Die Daten sind natürlich jetzt heute aktuell, da sie von der Version v2023.06.01 importiert und in die DB geschrieben wurden.

Die *.csv Dateien aus meinem Importverzeichnis schicke ich dir per Mail.

Vielen lieben Dank =)
Gruß Kai

ericplan, 2023-08-25:
Hello,Nothing to do with your current problem, but I guess you have a very old wrong database entry in tgeg_month on 7 and 8 may 2009. My guess this is from your first setup. Guy started ZonPHP. You can delete this value with PHPMyAdmin.Regards,Eric PlankeelVerstuurd vanaf mijn iPad

http://pv-scherbaum.de/index.php

seehase, 2023-08-25:
Good hint Eric... I have a supposition ... I will check tomorrow...

kds70, 2023-08-25:
> but I guess you have a very old wrong database entry in tgeg_month on 7 and 8 may 2009.

You´re right thank you. I will delete it ... ;)

seehase, 2023-08-25:
@kds70 kannst du mir vielleicht auch einen Datenbank Dump zuschicken, ich habe einen Verdacht und evt. kann ich einen workaround finden... bisher konnte ich den Fehler nicht nachstellen aber mit einer kompletten DB klappt das vielleicht

kds70, 2023-08-25:
@seehase ist unterwegs per Mail.

seehase, 2023-08-26:
Fehler gefunden und in der akutellen Entwicklungsversion 
https://github.com/seehase/ZonPHP/archive/refs/heads/feature/next_features3.zip
gefixed

Änderungen werden so schnell wie Möglich in das nächste Release mit aufgenommen
Bitte nochmal testen, ob es jetzt so klappt

closed 2023-08-26 by seehase

seehase, 2023-08-29:
Fixed with current release