seehase/ZonPHP #99: Import Sunny Explorer fails

tutunciyan, 2024-02-27:
I'm importing data by using the csv export files generated with Sunny Explorer.

The first data line of the export sometimes lacks the kW column value when the value is 0 kW.
Sample file snippet:

```
sep=;
Version CSV1|Tool SE|Linebreaks CR/LF|Delimiter semicolon|Decimalpoint dot|Precision 3

;SN: 2001209429;SN: 2001209429
;SB 2100TL;SB 2100TL
;2001209429;2001209429
;Totale opbrengst;Vermogen
;Counter;Analog
dd.MM.yyyy HH:mm:ss;kWh;kW
02.02.2023 08:50:00;22981.102
02.02.2023 08:55:00;22981.105;0.036
02.02.2023 09:00:00;22981.108;0.036
02.02.2023 09:05:00;22981.110;0.024
``` 

As you can see, the first line lacks the last kW value. Due to this, I'm getting the following error:

```
System.FormatException: Input string was not in a correct format.
   at System.Number.ThrowOverflowOrFormatException(ParsingStatus status, TypeCode type)
   at System.Number.ParseInt32(ReadOnlySpan`1 value, NumberStyles styles, NumberFormatInfo info)
   at System.Int32.Parse(String s)
   at MyNamespace.IntParser.Parse(String s) in C:\apps\MyNamespace\IntParser.cs:line 11
   at MyNamespace.Program.Main(String[] args) in C:\apps\MyNamespace\Program.cs:line 12
```

It needs an fail safe parsing with defaults probably.

seehausen, 2024-02-27:
Issue found and fixed:

please checkout latest version (v3.3.7)
or just replace line in sunny_importer

` $currentWatt = str_replace(',', '.', $lineValues[2]) * 1000;`

with
               ```
 $currentWatt = 0;
                $currentWattStr = trim(str_replace(',', '.', $lineValues[2]));
                if (strlen($currentWattStr) > 0 && is_numeric($currentWattStr)) {
                    $currentWatt = $currentWattStr * 1000;
                }
```

closed 2024-02-27 by seehausen

tutunciyan, 2024-03-21:
I'm still seeing errors in the above case:

> Fatal error: Uncaught TypeError: number_format(): Argument #1 ($num) must be of type float, string given in C:\inetpub\wwwroot\zon\importer\sunny_explorer.php:55 Stack trace: #0 C:\inetpub\wwwroot\zon\importer\sunny_explorer.php(55): number_format() #1 C:\inetpub\wwwroot\zon\importer\sunny_explorer.php(14): mapLinesToDBValues() #2 C:\inetpub\wwwroot\zon\inc\import_data.php(18): include_once('...') #3 C:\inetpub\wwwroot\zon\inc\connect.php(89): include_once('...') #4 C:\inetpub\wwwroot\zon\index.php(4): include_once('...') #5 {main} thrown in C:\inetpub\wwwroot\zon\importer\sunny_explorer.php on line 55

tutunciyan, 2024-03-21:
Content of the file where it fails on:

> sep=;
Version CSV1|Tool SE|Linebreaks CR/LF|Delimiter semicolon|Decimalpoint dot|Precision 3

>;SN: 2001209429;SN: 2001209429
;SB 2100TL;SB 2100TL
;2001209429;2001209429
;Total yield;Power
;Counter;Analog
dd.MM.yyyy HH:mm:ss;kWh;kW
21.03.2024 07:00:00;24864.840
21.03.2024 07:05:00;24864.840;0.000
21.03.2024 07:10:00;24864.840;0.000
21.03.2024 07:15:00;24864.842;0.024

seehausen, 2024-03-21:
Hi @tutunciyan 
Please check version 3.3.9
Should work, not sure who it worked in the past 
¯_(ツ)_/¯

seehausen, 2024-03-24:
@tutunciyan did the fix worked?
any feedback?

tutunciyan, 2024-03-25:
Hi, did an manual test and it seems to work 👍