seehase/ZonPHP #108: day_chart

ffi057, 2025-01-03:
Hello,
I installed ZonPHP v4.0.15 with my old database from ZonPHP, now my day_chart looks like this
[image]
the chart is only partly displayed.
When I look at the datebase i've got every 5 minut a meassurment.
What could't be wrong?

seehausen, 2025-01-03:
Is your installation public? can you send me a link to my private email (seehausen@gmx.de)
then I can have a look... did not see this before

ffi057, 2025-01-03:
https://demaeyerpower.be/index.php

snakepitnetwork, 2025-01-03:
Servus,Siehe www.snakepitnetwork.deGrüße Sepp

seehausen, 2025-01-03:
Bug found and fixed in version `v4.0.16`
changes can be seen here
https://github.com/seehase/ZonPHP/pull/109

Improvement of solarlog importer will follow

seehausen, 2025-01-03:
Please test the fix, if it works we can close the issue

ffi057, 2025-01-03:
By testing i saw that there is still a bug between 03/11/.. and 31/12/.. no data show for every year.

 

Nice regards,

 

Stefaan

seehase, 2025-01-04:
Fixed with version `v4.0.17`

closed 2025-01-04 by seehase