seehase/ZonPHP #115: import .js log files

Pdw112, 2025-03-14:
Hi,

Is there an option to import my .js logfiles.
this are the filename of my logfile: min250314.js
Data in the file are:
m[mi++]="14.03.25 11:20:00|1784;1951;2120;220;33"
m[mi++]="14.03.25 11:15:00|1108;1213;2050;230;30"
m[mi++]="14.03.25 11:10:00|1546;1690;1900;238;32"
m[mi++]="14.03.25 11:05:00|1445;1576;1810;225;26"
m[mi++]="14.03.25 11:00:00|628;681;1680;243;24"
m[mi++]="14.03.25 10:55:00|301;324;1560;243;25"
m[mi++]="14.03.25 10:50:00|703;762;1510;243;26"
m[mi++]="14.03.25 10:45:00|250;269;1480;236;27"
m[mi++]="14.03.25 10:40:00|1132;1228;1420;235;29"
m[mi++]="14.03.25 10:35:00|842;914;1400;230;28"
m[mi++]="14.03.25 10:30:00|777;845;1310;240;31"
m[mi++]="14.03.25 10:25:00|1281;1397;1240;224;31"
m[mi++]="14.03.25 10:20:00|1298;1416;1180;233;35"
m[mi++]="14.03.25 10:15:00|1572;1721;1060;225;35"
m[mi++]="14.03.25 10:10:00|1548;1694;960;232;34"
m[mi++]="14.03.25 10:05:00|1520;1662;820;231;34"
m[mi++]="14.03.25 10:00:00|1498;1636;700;232;33"
m[mi++]="14.03.25 09:55:00|1500;1640;570;232;33"
m[mi++]="14.03.25 09:50:00|1496;1635;440;233;33"
m[mi++]="14.03.25 09:45:00|1482;1619;320;235;33"
m[mi++]="14.03.25 09:40:00|1468;1605;190;235;33"

Now I use the super old zonphp from slaper to show my values.

seehausen, 2025-03-14:
Just to get it right. 
you want to always import your date from  .js files
so your export data from your converter into *.js, and use this data to import into ZonPHP?
or should it be a one time import only?

If you have an old version, you should be able to upgrade to the new (current) version without problem, 
database structure is still the same, and migration is done on the flie

Currently there is no "importer" for *.js files, but I can implement one if needed.
That I can do, if you switch to the current version and you always have data in this format to import

seehausen, 2025-03-20:
@Pdw112 Can you provide more information

Pdw112, 2025-03-22:
I overwrite the old files.
It reads the old data but not de new data.
See.
[](http://weerstationtilburgreeshof.nl/zon/)

seehausen, 2025-03-22:
Hi @Pdw112 

If you do not specify an importer, ZonPHP cannot import any new data.

I still did not get your configuration
You have a sub-folder where you have all your `min250xxx.js` files, so one file per day?
do you have a chance to export the data in csv format? Or can our only use the *.js format

Please send my as an example all *.js files you have from this month (all jsons from march)
to my private email: seehausen@gmx.de

And tell me, where the files are located. `HTML_PATH: /zon/<my data sub folder>`  in which subfolder of you ZonPHP installation

seehausen, 2025-04-15:
Sorry I did not get the message. can you resend it to me -> seehausen@gmx.de

Pdw112, 2025-04-16:
Resend to your mailbox

seehausen, 2025-04-26:
Checked the js files
unsure what the values in the files means

```
m[mi++]="14.03.25 10:05:00|1520;1662;820;231;34"
m[mi++]="14.03.25 10:00:00|1498;1636;700;232;33"
```
for a 5min record I see 5 values
How to calculate or get the produced kWh in that 5min range.
Actually I need to know only that value e.g 1350W average for 5min
or also possible: 15.53kWh produced in the last 5min

Pdw112, 2025-04-27:
m[mi++]="date time|W to net;W totall pannel side;Kwh produced that day;V;temp"
m[mi++]="14.03.25 10:00:00|1498;1636;700;232;33"

seehausen, 2025-05-13:
Sorry I took a bit longer
the current version 4.2.0 supports the import of js files

please
set 
`importer = import_json`
in you parameter.php file

If you have any problems, please reach out to me

Pdw112, 2025-05-15:
Thanks.
I will update asap.

Verzonden vanaf Outlook voor Android<https://aka.ms/AAb9ysg>